bakkerthehacker/serqet #36: Find a good PCB manufacturer

kurtmh, 2015-01-14:
Links I've gathered:
http://www.4pcb.com/
http://www.seeedstudio.com/service/index.php?r=pcb
https://oshpark.com/ 
Word on the street is "Pretty sure all 3 will apply solder mask, no stencils tho( if you're designing anything that's leadless) and will do e-tests for free. "

http://volterainc.com/ Word on the street is " if you know tron/mech people, they're beta testing their printer for 10 bux/board" "Apparently Voltera doesn't have a drill/pick and place attachment, so no holes/vias or assembly help if you're using that" "talk to alroy from voltera in waterloo"
http://imall.iteadstudio.com/open-pcb/pcb-prototyping.html

kurtmh, 2015-01-14:
Our current approximate guide for pricing is <$200 for 20 boards.

kurtmh, 2015-01-14:
http://www.apcircuits.com/ recommended in course notes.

kurtmh, 2015-01-15:
On Campus: https://uwaterloo.ca/3d-print-centre/circuit-board-manufacturing. I went today and asked about rough pricing and got a figure of ~$16/board depending on complexity. It's the same price per board for additional boards. They don't do verification but they said they haven't heard of issues in a while.

kurtmh, 2015-02-16:
Oshpark.com can be used for a quick quote. Carl says they take about 2 weeks. There is crimp circuits in Toronto which could take much less time but will cost more. There's also still the school. Carl says there's 3 more places in Waterloo.

aadavids, 2015-03-16:
done

closed 2015-03-16 by aadavids